jonaslorander/OctoPiPanel #17: video system not initialized

tiagojanz, 2015-01-04:
Can't run 

After I issue the command sudo python ./OctoPiPanel.py  I get

Traceback (most recent call last):
  File "./OctoPiPanel.py", line 595, in <module>
    opp = OctoPiPanel(320, 240, "OctoPiPanel!")
  File "./OctoPiPanel.py", line 135, in __init__
    pygame.mouse.set_visible(False)
pygame.error: video system not initialized

jonaslorander, 2015-01-04:
I need information; what display are you using? Can you see the console on the display? Are you using OctoPi or have you installed OctoPiPanel using the information in readme? Are you using master or devel branch?

tiagojanz, 2015-01-04:
Hi,

Thank you by the response

Im using a 7" egalax display

I can see everything in the display, it works like a ordinary display with
added bonus a touchscreen.
Im using OctoPi image
Octoprint is devel, but test it as master as well with same result.

tiagojanz, 2015-01-04:
I can give you ssh remote access if you want.

jonaslorander, 2015-01-04:
I am on vacation and will help you out as well as I can witbout SSH :)

I know nothing about this display. Is it connected with HDMI? If so, you can try to change '/dev/fb1' to /dev/fb0' around line 125, and see if that helps. Or commenting the line altogether.

If it runs over the GPIO header with SPI I can't help you now...

jonaslorander, 2015-01-04:
'/dev/fb0' (forgot the first ' )

tiagojanz, 2015-01-04:
Hi,

Its connected by Hdmi, tried that but still doesn't work, I tried also to
change from 320 x  240 to 640

jonaslorander, 2015-01-04:
Hum. Try removing all four lines around /dev/fb1, it should be possible to get it to work. You could also try starting OctoPiPanel from within X.

tiagojanz, 2015-01-06:
It did work to comment 

       # if platform.system() == 'Linux':
           # Init framebuffer/touchscreen environment variables
           # os.putenv('SDL_VIDEODRIVER', 'fbcon')
           # os.putenv('SDL_FBDEV'      , '/dev/fb0')
           # os.putenv('SDL_MOUSEDRV'   , 'TSLIB')
           # os.putenv('SDL_MOUSEDEV'   , '/dev/input/touchscreen')

now its working.

I will be tinkering to get it more 7 inch friendly.

by the way GREAT WORK!!

jonaslorander, 2015-01-06:
Great to see that you got it working :)

closed 2015-01-06 by jonaslorander