jonaslorander/OctoPiPanel #18: Enlarge view area for 7"

tiagojanz, 2015-01-06:
Hi,

It does work great as is, but... I have a big screen and theres so much space there:)

I have tried to expand the view area

  def __init__(self, width=640, height=480, caption="OctoPiPanel"):

and added 2 more buttons

        self.btnXL           = pygbutton.PygButton((245,  5, 100, self.buttonHeight), "X -")
        self.btnXR           = pygbutton.PygButton((245,  35, 100, self.buttonHeight), "X +")

but the buttons don't show up fully they are right cutted,  how can I expose more canvas?

tiagojanz, 2015-01-07:
Heres a screenshot

[image]

jonaslorander, 2015-01-17:
Interesting issue. I will take a look at it and see if I find something.

jonaslorander, 2015-01-17:
You need to check you math :)
The X-position of the buttons hould increase with button width + button space (100 + 5 = 105), so the buttons should not have X-position of 245, but 320.

This works:
```
self.btnXL            = pygbutton.PygButton((320,  5,  100, self.buttonHeight), "X -")
self.btnXR            = pygbutton.PygButton((320,  35, 100, self.buttonHeight), "X +")
```

[image]

closed 2015-01-17 by jonaslorander

tiagojanz, 2015-01-18:
You are right !!!!
Do you have plans for keyboard integration? I already have some code.

jonaslorander, 2015-01-18:
Keyboard integration is beyond the scope of this project. It is supposed to be small enough be mounted on the printer, without the need for a keyboard or a mouse.