jonaslorander/OctoPiPanel #21: Running OctoPiPanel in LXDE

sudpuzzer, 2015-01-23:
I've been having some trouble getting good results with OctoPiPanel. 
I am using a resistive touchscreen overlay on a 7inch LCD printer-mounted display attached to my RPi via HDMI. The touchscreen overlay uses an AR1100 controller, which Raspbian picks up as a USB mouse (event0). By commenting the SDL_MOUSEDEV and SDL_MOUSEDRV lines, and changing the frame buffer to fb0, I can get OctoPiPanel to come up on my LCD and use the touchscreen. However, the mouse calibration isn't right in pygame. I've calibrated the AR1100 controller with Windows 7, and when I start an X session, the mouse calibration is perfect in LXDE.

Is there any way to run OctoPiPanel in a window in LXDE, or something I can investigate to correct the mouse mapping?

jonaslorander, 2015-01-23:
Hi,

Try commenting out this entire block of code (line 94-99):
```
        if platform.system() == 'Linux':
            # Init framebuffer/touchscreen environment variables
            os.putenv('SDL_VIDEODRIVER', 'fbcon')
            os.putenv('SDL_FBDEV'      , '/dev/fb1')
            os.putenv('SDL_MOUSEDRV'   , 'TSLIB')
            os.putenv('SDL_MOUSEDEV'   , '/dev/input/touchscreen')
```
It could work. I have not tested it at all since I don't have LXDE up and running anywhere...

sudpuzzer, 2015-01-23:
That absolutely worked. I commented out that block and ran it using

`sudo python OctoPiPanel.py`

in the LXDE terminal. This gave me a window in X where the mouse works perfectly. Now I just have to resize everything for my 1280x800 window and I'll have a nice little control panel.

guysoft, 2015-01-24:
Suggest either one of you makes a pull request with something along the lines of:

    import subprocess
    
    if platform.system() == 'Linux':
        if subprocess.Popen(["pidof", "X"], stdout=subprocess.PIPE).communicate()[0].strip() == "" :
                # Init framebuffer/touchscreen environment variables
                os.putenv('SDL_VIDEODRIVER', 'fbcon')
                os.putenv('SDL_FBDEV'      , '/dev/fb1')
                os.putenv('SDL_MOUSEDRV'   , 'TSLIB')
                os.putenv('SDL_MOUSEDEV'   , '/dev/input/touchscreen')

This would automatically detect if Xorg is running and load accordingly

closed 2015-01-27 by sudpuzzer

bolsoncerrado, 2015-02-19:
Im getting the following no matter if I load it on the login shell or the LXDE terminal window:

pi@octopi ~/OctoPiPanel $ sudo python OctoPiPanel.py
Traceback (most recent call last):
  File "OctoPiPanel.py", line 11, in <module>
    import requests
ImportError: No module named requests
pi@octopi ~/OctoPiPanel $ 

any clues?

Thanks!

jonaslorander, 2015-02-19:
Yeah, read the README.md again and install the requirements ;)

```
cd ~/OctoPiPanel
sudo pip install -r requirements.txt
```

bolsoncerrado, 2015-02-19:
Yeah saw it right after hitting COMMENT, sorry! :D

OK i have it running now ;)

Final question, what would be the best way to get a reasonable sized button set and such? ie, optimized for my resolution? where are the buttons created? I understand this is not HTML right?

What about that other "Panel" someone was posting in redofthedead in the other comment? ;)

Thanks!

jonaslorander, 2015-02-19:
https://github.com/jonaslorander/OctoPiPanel/blob/devel/OctoPiPanel.py#L127-142 are the lines that created the buttons.