jonaslorander/OctoPiPanel #24: Buttons Not Working

eboston, 2015-01-26:
I am having an issue with the screen buttons not working.  I have put print statements in the code and I see the button getting created and when clicking on the screen, a mouse click event.  However, when clicking on any button, the button click does not register but I do get a mouse click event.

Running dev version of OctoPrint and the latest OctoPiPanel code from the dev branch.

jonaslorander, 2015-01-27:
Can you check your OctoPiPrint.cfg and see if the `baseurl`- and `apikey`-properties ar correct?

eboston, 2015-01-27:
The baseurl is http://localhost:5000 and the apikey matches.  But since the code never reaches the part where they are needed, the button handling code is never called, I'm not sure where those would be an issue yet.

jonaslorander, 2015-01-27:
That is odd. Never heard of anything like it before. 

Do you enter the if-clause on line 218 or 250, or both? Do you have a maximum time for the background to turn off, or is it on all the time?

eboston, 2015-01-27:
I get the same behaviour regardless of the background timeout setting.  With a non-zero value, the screen blanks but turns back on when I touch the screen.  

I added the print statements around line 218

           if self.bglight_on == True:
                print "bglight is on"
                if 'click' in self.btnHomeXY.handleEvent(event):
                    print "Home X/Y button clicked"
                    self._home_xy()

and 250
            # Did the user click on the screen?
            if event.type == pygame.MOUSEBUTTONDOWN:
                # Reset backlight counter
                print "Mouse click"
                self.bglight_ticks = pygame.time.get_ticks()

Output when pressing the pause button...
(I also added the extra print statements shown at startup)

pi@octopi ~/scripts $ ./OctoPiPanel
Starting...
Home X/Y button created
OctoPiPanel initiated
OctoPiPanel started!
---
bglight is on
bglight is on
Mouse click
bglight is on
bglight is on
bglight is on
bglight is on
bglight is on
bglight is on
bglight is on
bglight is on
bglight is on

eboston, 2015-01-27:
I added some trace statements in the pygbutton.py and I am not seeing it ever handling any clicks.  How do I determine if I have the correct pygame installed?  I am using the one that came with the distribution.

jonaslorander, 2015-01-27:
It is supposed to work with the Pygame supplied with Raspbian. I will make some test when I get home from work and see if I can make sense of it.

jonaslorander, 2015-01-27:
Stupid question, but I assume you have calibrated the touchscreen?

eboston, 2015-01-27:
There were no directions on calibrating the display from adafruit. I did try running the rpi calibration but it reported iy did not recognize the display.

sudpuzzer, 2015-01-27:
You can try calibrating with TSLIB. In the console:

    export TSLIB_TSDEVICE=/dev/input/touchscreen
    export TSLIB_FBDEVICE=/dev/fb1

    ts_calibrate 

You can test your calibration using

`ts_test`

eboston, 2015-01-27:
Will give that a try after work. These instructions should probably be added to the readme.

jonaslorander, 2015-01-27:
The calibration shouldn't really be needed anymore after Adafruit updated the install script. And I assume when using Adafruit PiTFT, that it has been installed properly before attempting anything else.

eboston, 2015-01-27:
sudpuzzer, I was able to run the calibration and test.  But the buttons are still not responding in OctoPiPanel.

jonaslorander, 2015-01-28:
@eboston Have you ever tried running X on the PiTFT?
You can start X on the PiTFT using this:
`FRAMEBUFFER=/dev/fb1 startx`
You can't be running X on any other source, like HDMI in the same time.
Then you would be able to see if the screen and touch is working properly. It could be that the touch is inverted or something.

eboston, 2015-01-28:
@jonaslorander I was able to run X just fine, other than it being very slow.  but was able to click different places on the screen and get the correct response.

eboston, 2015-01-28:
@jonaslorander One other thing I noticed, and probably not completely related, is I tried setting up the script for automated start up but it doesn't work.  Thinking something may be corrupted on my system, but I have installed it twice so far with similar results.

eboston, 2015-01-28:
@jonaslorander Okay.  Think I am getting closer.  When I was testing out the X, I realized I had left the config setup to boot to the GUI.  I turned that off and was able to run X on the screen just fine.  After that, when I tried running OctoPiPanel again, the buttons started working, sort of.  Alignment seemed off so I ran the ts_calibrate.  I had been able to get the Z + 25 button to actually do something.  But now after running the calibration, when I clicked a button (not sure which one as there was a lag between the time I clicked it and the action happened), the RPi rebooted.  So it looks like the button clicks are getting through now, just not at the right location.

jonaslorander, 2015-01-29:
I had problems to with not getting it to autostart. The problem was that there were Windows line endings in the file. So, if you have edited any .py files on Windows and then uploaded them to your Pi that could be the problem.

Regarding the buttons, it seems that the display is not properly calibrated. When you clicked around in X on the PiTFT, were the clicks in the right place?

If you have used this DIY-script you should be all set without the need calibration:
https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install#diy-installer-script

eboston, 2015-01-29:
@jonaslorander Well, running that script seemed to have resolved the calibration issue.  Going to have to try setting up a clean install and record all the steps that I had to do to get it working.  Thanks for the help.

jonaslorander, 2015-01-29:
Glad to be of help.
I will update the README to inlcude the link to how to setup the display.

closed 2015-02-14 by jonaslorander