jonaslorander/OctoPiPanel #28: won't start as daemon

jneilliii, 2015-02-23:
Where is the first place to look in order to debug display issues with OctoPiPanel?  I followed the instructions for installing on the main readme page as a daemon to auto-start but the only thing displayed on my touch screen is the console.  I can open startx and the touchscreen is working properly and displaying.  Do I have to login to the console or start some other interface to get it to load?

jonaslorander, 2015-02-23:
It should start automatically if you have done it correctly. Did you run ```sudo update-rc.d octopipanel defaults```? Also, can you start it by typing ```sudo service octopipanel start``` from an SSH-connection? Double check that OctoPiPanel.py has the executable attribute. Have you edited any of the files on a Windows-machine, and accidentally changed line-endings?

guysoft, 2015-02-23:
BTW I just stick in ``/etc/rc.local``:
``/home/pi/scripts/OctoPiPanel``. Didn't know that this option is there, is not documented anywhere.

nophead, 2015-02-23:
Have you tried running it from the command line first? That will show any
errors which cause it to terminate immediately when run as a service.

jonaslorander, 2015-02-23:
@guysoft https://github.com/jonaslorander/OctoPiPanel#automatic-start-up ;)

jneilliii, 2015-02-24:
So got around to working on this again tonight and running the command in an ssh console gives error below.  Do I need to run this command locally on the Pi with a keyboard?
```
Traceback (most recent call last):
  File "OctoPiPanel.py", line 575, in <module>
    opp = OctoPiPanel("OctoPiPanel!")
  File "OctoPiPanel.py", line 108, in __init__
    pygame.mouse.set_visible(False)
pygame.error: video system not initialized

```

jneilliii, 2015-02-24:
And for some additional background information, I used [this](http://eleduino.com/products_ny_137.html) lcd touchscreen and followed [these](http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/) directions to get it running on my raspbian image, or at least to get the console to show up...

jonaslorander, 2015-02-24:
It will run natively without X or keyboard.

These are the lines that are of interest for you:
https://github.com/jonaslorander/OctoPiPanel/blob/devel/OctoPiPanel.py#L98-101

You have to make sure that the correct framebuffer device is being used, and the correct framebuffer driver.

You could try running this script to see if anything of that works, and you can get any video from Pygame.
https://learn.adafruit.com/pi-video-output-using-pygame/pointing-pygame-to-the-framebuffer

jonaslorander, 2015-02-24:
Also, make sure you don't boot into X before starting OctoPiPanel (to be on the safe side).

jneilliii, 2015-02-25:
Thanks for those pointers.  After tryng the pyscope.py test nothing was working which led me to downloading and compiling the necessary C libraries for bcm2835 and WiringPi.  Once I did that and rebooted the pyscope.py script now displays the red box over the whole screen and displays the mouse in the upper left corner.  It returns "Framebuffer size: 320 x 240" and then closes as expected.  Unfortunately, this did not resolve my issue with OctoPiPanel.  I still get the same error message as before.

jonaslorander, 2015-02-25:
Then I would guess the screen uses fbcon, since that is the first in the list in pyscope and it did not report that a driver did not work.
The setup of the display is different i pygame then it is in OctoPiPanel. Perhaps you could make changes around L111 to match those of pyscope.

jneilliii, 2015-02-26:
After mucking around with this for  a while I was able to figure out that my hacked together raspbian image wasn't going to work, or at least I didn't want to spend the time figuring it all out.  Ended up using  an img file provided by the eleduino company and rebuilding my octoprint server from scratch, leading to having to recompile the arduino code to revert back to a standard baud rate for serial communication, etc.  

Long story but it appears that might have been the trick as OctoPiPanel is now running correctly, albeit the entire display is rotated 180 degrees from how I have it mounted on my printer.  For the life of me I can't get it to rotate correctly and stop X from loading on boot.

jneilliii, 2015-02-26:
Take that back.  I am able to run the script from the command line, but running as a daemon still seems broken.  The screen goes blank and the only thing displayed is a white underline in the what would be lower right corner (based on correct orientation to UI).

jneilliii, 2015-02-26:
Thanks to the hint from guysoft I now don't have LXDE starting on boot and OctoPiPanel is loading from the rc.local file instead of from the daemon.  My guess is that the errors listed previously are preventing the daemon from running properly.

jneilliii, 2015-02-26:
Ok, so I take that back.  It works fine as a daemon as long as the LXDE is not running on startup.

closed 2015-02-26 by jneilliii

jonaslorander, 2015-02-26:
I'm happy it worked out for you!

jonaslorander, 2015-02-26:
Oh, and the errors you printed has to do with turning the backlight off on a few other LCDs. And you LCD donät seem to use these GPIOs so thats why it fails (cause they don't exist). I'm thinkin of making a check if they exists before accessing them, but thats for the future :)