jonaslorander/OctoPiPanel #30: Backlight not turning off.

DaveyRance, 2015-05-05:
No matter what i do it seems that my backlight will not turn off on the display.

I have checked the OctoPiPanel.cfg file and it has the default timeout of 30 seconds but it seems that it is just getting ignored by my system.

The display auto starts and works fine but just will never turn off so it is getting the non-printing screen burnt in.

My version of OctoPi is 1.1.1-30-g4fede5a (master branch)
From the README i am running OctoPiPanel v0.1 

It is on a PI 2.0.

Any ideas on what to check?

nophead, 2015-05-05:
No idea, but does turning the light off really prevent screen burn? I think you need to erase the image, or move it about.

jonaslorander, 2015-05-06:
What LCD have you got? I only have the Adafruit PiTFT on hand and rely on the community to update OctoPiPanel to work good with other LCDs.

DaveyRance, 2015-05-06:
The version one of the adafruit

jonaslorander, 2015-05-06:
That's odd then. Are you running devel och master branches? Are you running OctoPiPanel using sudo? If you are running devel, you could try adding a print statement inside the if clause on line 174 and see if the code executes. So we now if it is a software or hardware/driver/Setting related problem.

stirobot, 2015-09-18:
Same issue.  Raspi 2, octopi image (05-05 build), resistive 2.8" adafruit official pitft.  Installed using the kernel stuff from adafruits automated instructions.

martinreinhold, 2015-09-26:
I think I may have the source of this issue figured out.   It looks like newer versions of the adafruit kernel patch uses a reference to pin 508 instead of 252 to control the backlight.  

https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/backlight-control

jonaslorander, 2015-09-28:
Good eye! I'll fix that. @DaveyRance Can you see if it works with the new code?

DaveyRance, 2015-10-03:
@jonaslorander Works perfectly now :) now after the time set in the cfg file the display goes nice and black.

closed 2015-10-03 by DaveyRance