jonaslorander/OctoPiPanel #31: Update OctoPIPanel

Helisinus, 2015-05-06:
i know this is not an issue, but i don't know where i can ask this question.How can i update my "old" OctoPIPanel version? 

Thanks for your help!
Joerg

jonaslorander, 2015-05-06:
Stop OctoPiPanel and the issue this command to update:
```
cd ~/OctoPiPanel/
git pull
cd OctoPiPanel
sudo pip install -r requirements.txt
```

If you would like to switch branch to devel, do this:
```
cd ~/OctoPiPanel/
git pull
git checkout devel
cd OctoPiPanel
sudo pip install -r requirements.txt
```

Then you can start OctoPiPanel again.

jonaslorander, 2015-05-06:
I updated the readme file for the devel branch as well. :)

Helisinus, 2015-05-06:
Yes, this works!!
Thanks for your fast help!!

Joerg

closed 2015-05-06 by jonaslorander