jonaslorander/OctoPiPanel #34: Octopi Panel quits when connecting to the printer.

mwalimu, 2015-07-21:
I'm finding that octopi panel quits whenever I connect the printer in octoprint.  It seems to work for a brief second, but then quits.

guysoft, 2015-07-21:
What printer?
What version of OctoPi?

jonaslorander, 2015-07-21:
What happens if you connect the printer to OctoPrint before starting OctoPiPanel. During development I'm running OctoPiPanel on a model B RPi, which has 512 MB RAM. Model A+ only has 256 MB RAM. But I think that should be enough.

mwalimu, 2015-07-21:
The most recent version of octopipanel,  a printrbot simple metal and M3D printer.  I just checked, it works fine on the B+, but A+ quits when I connect to the printer.

guysoft, 2015-07-21:
How did you get OctoPrint to work with an M3D priner?
BTW I actually got one here, but it broke down when trying to follow the FAQ to fix the internal filament feed (I need to message M3D about it, I hope their warrenty covers faulty FAQ instructions).

jonaslorander, 2015-07-21:
Perhaps I should buy a Model A+ for future development. Are you using dev branch (I should really release it)?

mwalimu, 2015-07-21:
see this thread for Octoprint on M3D.

https://printm3d.com/portal/forum/index#/discussion/1784/linux-software

It's not perfect, but it does work.

mwalimu, 2015-07-21:
I thnk I'm using the regular branch(?).  Basically I'm following the README directions.

jonaslorander, 2015-07-21:
You should use the dev branch. M sure you use it and see if the problem persists. But I can't really think of anything that would make it crash just because if you running it on Model A+. Does it say anything on Adafruit about running the display on Model A+? I'm just assuming you are using a screen from adafruit.

mwalimu, 2015-07-22:
Tried the dev branch.  Same results.  Yes it is an Adafruit Pitft 2.8.  That runs just fine on the Pi A+.

jonaslorander, 2015-08-17:
I purchased a Model A+ the other day. When I get the time I'll try to reproduce you error.

closed 2021-01-18 by jonaslorander