jonaslorander/OctoPiPanel #35: Octopipanel crashes

olovfr, 2015-09-21:
Hello,
I have tried for two days to get it to work. I have tried both on RPi2 and B+.
I have managed to get octopipanel to start but it doesn't seem to be connected as the temps ar not right (0 degrees). When I press a button octopipanel exits. In commandprompt it says something about "temp" I run from octopi master. I have tried octopipanel level and master branch.  
Is it a setting in octroprint I have missed? I copied the API code from settings interface. I have tried with localhost and with octoprint ip adress.

jonaslorander, 2015-09-21:
Hi olovfr,

It sound like you are running a old version of OctoPrint. Which version are you using? There are a few changes in the REST API regarding temperatures between 1.1x to 1.2 on OctoPrint.

olovfr, 2015-09-21:
Hello, I tried to install the latest Octopi images and octoprint was upgraded to the latest.
Unfortenely it didnt work anyway. Still KeyError: 'temps'.

nophead, 2015-09-21:
That should have been fixed by my change here: https://github.com/jonaslorander/OctoPiPanel/commit/7ca9b93da81c8c57a6495a8399f331129e2adc33

It looks like you don't have that change.

jonaslorander, 2015-09-22:
@olovfr Could you try updating OctoPiPanel with theese instructions: https://github.com/jonaslorander/OctoPiPanel#update-octopipanel
I don't know which commit of OctoPiPanel OctoPi has in the latest release. Perhaps @guysoft knows?

olovfr, 2015-09-23:
I have updated and switch t "devel" now it works but crashes after reboot. When I start the service again manualy it works.

olovfr, 2015-09-23:
I get 
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/mode: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/frequency: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/duty: Directory nonexistent

stirobot, 2015-09-24:
I used this: https://github.com/jonaslorander/OctoPiPanel/commit/7ca9b93da81c8c57a6495a8399f331129e2adc33 
and now it works

jonaslorander, 2015-09-24:
@olovfr Those errors are to do with the background light, turning it on and off. As there are many different displays out there that don't use the same pins for controlling the backlight we will have several different ways of controlling it and some will not work. This error has never, to my knowledge, caused any problems.

What display are you using? I see you havn't mentioned that and we all have assumed you use one from Adafruit or Watterott that have been tested to work.

olovfr, 2015-09-24:
Thank you for your help. I have an Pitft from Adafruit. Maybe it´s not that error that makes it crash. Yesterday it started as it should but this morning after restart of the Rasberry it crashed and I manually had to restart the service. 
I have tried different octopi images now I run 2015-07-02_2015-05-05-octopi-wheezy-0.12.0.zip
I manually updated octoprint and then the screen started to receive information. But sometimes it crashes at start up and sometimes it starts.  I whist there was an an ready to go image with adafruit pitft drivers and octopi for RBPi2 because the readymade image from adafruit seam to work but I had difficulties to install octoprint and fix it as good as Octopi.

To make it clear.
1. I installed the 2015-07-02_2015-05-05-octopi-wheezy-0.12.0.zip
2. I run the https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install#diy-installer-script)
3. I manually updated Octoprint by command line.
4. I run the update of octopipanel and changed to "devel".

As it still crashes at startup I installed "adafruit occidentalis" as they recomended.

Now it works sometimes.

Did I do something wrong?

nophead, 2015-09-24:
Seems like you might have two versions of OctoPiPanel on your RPI. If you get KeyError: 'temps' you are running a very old version as that error is not possible for a long time.

olovfr, 2015-09-24:
Okej,
Strange. I followed the guide from https://github.com/jonaslorander/OctoPiPanel to install it.
Dosn´t it install the latest version? I also manually updated as suggested. How do i make sure I run one version and the latest?

olovfr, 2015-09-24:
"I used this: 7ca9b93" I will try a fresh install with this. Im not shure how to change to this file but I hope to work it out.

jonaslorander, 2015-09-24:
If you have the latest OctoPi you can run the script found in /home/pi/script
enable-adafruit-pitft installs PiTFT and then you can run OctoPiPanel with OctoPiPanel script in the same catalog. It will even get the correct API-key for you.

So after you start OctoPi for the first time, run enable-adafruit-pitft, then reboot and run OctoPiPanel.

olovfr, 2015-09-24:
I will try that again. Maybe I didn´t have the latest release last time I tried. Because when I first tried it wasn´t able to boot after. Thank you again!

jonaslorander, 2015-09-24:
Tell us if it works for future reference.

olovfr, 2015-09-24:
Hello Im sorry to say that even in 2015-07-02_2015-05-05-octopi-wheezy-0.12.0.zip Octopi dosn´t boot after running the script "enable-adafruit-pitft". It may be because im running Raspberry Pi 2.

jonaslorander, 2015-09-24:
Could be. I have not tried this on Raspberry Pi 2.

@guysoft Have you seen any problems iwhen running OctoPi with SPI LCD on RPi2?

nophead, 2015-09-24:
IIRC the problem is the adafruit script loads a kernel with TFT support but it is too old for the Pi 2.

olovfr, 2015-09-24:
nophead I think you are right it works when I load the kernel from adafruits page to boot the Pi but then i have problems with Octopipanel.

olovfr, 2015-09-24:
Right now Im trying with a nightly build of Octopi and hope that will fix it.

olovfr, 2015-09-24:
That did not work either so I dont think its possible to run the script at octopi if you have a RPI2.

nophead, 2015-09-24:
See https://github.com/guysoft/OctoPi/issues/72, it should be fixed now though.

jonaslorander, 2015-09-24:
At this point I suggest you install a clean Raspbian on your Pi. And then use the guides on Adafruit to get the screen working. Perhaps there is one for getting PiTFT working on RPi2? After that you install OctoPrint and then OctoPiPanel.

olovfr, 2015-09-24:
Yes I tried that yesterday but I wasn´t able to install octoprint and make it as good as octopi:(
Thank you all for your effort. I will try some more or get an Raspberry B+. Maybe a combination of your recomendations will work finaly.

olovfr, 2015-09-24:
Switched to Rasberry B+ and now it works. So the problem is with Rasberry Pi 2. Even with adafruits drivers.

closed 2015-09-25 by jonaslorander

Eddiiie, 2017-02-15:
Bump.. Any chance of getting this to work?  I have same problem.

jonaslorander, 2017-02-15:
The `KeyError: 'temps'` is almost always because the OctoPiPanel version you are using are not suited for the specific OCtoPrint version you are using.

If you are using OctoPrint newer then 1.1 you need to use the devel version of OctoPiPanel.

Eddiiie, 2017-02-15:
Thank you.    

I try to change to devel  but  dont think it works -

pi@toyrep:~/OctoPiPanel $ git checkout devel
error: pathspec 'devel' did not match any file(s) known to git.
pi@toyrep:~/OctoPiPanel $ sudo pip install -r requirements.txt
Requirement already satisfied: requests in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 1))

jonaslorander, 2017-02-15:
Try doing a `git pup` before `git checkout devel`.
Remember to save you config file first somewhere else then in the OctoPiPanel catalog as it might be overwritten.

Eddiiie, 2017-02-15:
git pup is not a recognized command.

then it asks if I meant pull  or push

jonaslorander, 2017-02-15:
Sorry my bad, it should be `git pull`.

Eddiiie, 2017-02-16:
Hi.  :)

```
pi@toyrep:~/OctoPiPanel $ git pull
Already up-to-date.
pi@toyrep:~/OctoPiPanel $ git checkout devel
error: pathspec 'devel' did not match any file(s) known to git.
pi@toyrep:~/OctoPiPanel $

```

Is it just me?

guysoft, 2017-02-16:
Try:
 git fetch origin

Eddiiie, 2017-02-18:
> 

> Try:
>  git fetch origin

pi@toyrep:~/OctoPiPanel $ git fetch origin
pi@toyrep:~/OctoPiPanel $

jonaslorander, 2017-02-20:
After ´get fetch origin´ you should do `git checkout devel`.