jonaslorander/OctoPiPanel #37: Only white screen

sysshad, 2016-01-08:
Hi,
Im trying to get this tft 3.5"  320x480 to work:
http://www.ebay.com/itm/281778730873?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

All i get though is a white screen.
i followed the steps shown on adafruit:
https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install
i chose y on both questions then rebooted. And after this is when i got the white screen.
Before that i had a black screen with white lines on it.

i then tried to follow this guide:
https://github.com/jonaslorander/OctoPiPanel
Also added the API key that i found on my octoprint.local and settings tab.
No help there.

Sorry, im a newbie at linux i could use all help thats possible.

jonaslorander, 2016-01-09:
Are you able to get anything to show up on the screen? Or just not OctoPiPanel?
I belive your problem is that the screen is not the same as the Adafruit one, and hence requires other drivers. And can't be installed using Adafruit tutorial... I have never seen this LCD Before, but surely there must be a guide on the Internet.

Perhaps this could be of help: https://www.raspberrypi.org/forums/viewtopic.php?f=93&t=79120

If you get your screen to work properly I can help you further with OctoPiPanel.

sysshad, 2016-01-09:
Ok thank you for your answer. Im going to try and find drivers.
My linux knowledge is very loe. But i think i got an error upon bootup when
the adafruit drivers tried to init the tft.
Brgs stig

sysshad, 2016-01-09:
I talked to the LCD supplier which in turn sent me a big ZIP file ZC59400.zip.
In this zip file there was an image (LCD35-141202.img) which i sent to SD card.
Upon first boot the LCD worked in an instant :) it even autostarted startx

I managed to read some fast information at startup that the display was ADS7846
and there was a folder in home dir named bcm2835-1.36.

Now, as a complete linux newbie, i really wonder how im going to get this working on octoprint image lol. I tried to install octoprint onto this image i had working for the LCD, but got tons of warnings and errors so i gave up :P

closed 2021-01-18 by jonaslorander