jonaslorander/OctoPiPanel #38: OctiPi Panel crashes with KeyError

SaftpresseWust, 2016-01-16:
Hello,
I have just installed OctoPiPanel. When I start it without OctoPrint running, the screen stays open. But when I start OctoPiPanel with Octoprint already running I get the following error:
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
OctoPiPanel initiated
OctoPiPanel started!
---
Traceback (most recent call last):
  File "./OctoPiPanel.py", line 596, in <module>
    opp.Start()
  File "./OctoPiPanel.py", line 194, in Start
    self.get_state()
  File "./OctoPiPanel.py", line 293, in get_state
    self.HotEndTemp = state['temps']['tool0']['actual']
KeyError: 'temps'

The OctoPiPanel Screen flashes open for a moment and shuts down immediatly.
I appreciate any Ideas.

jonaslorander, 2016-01-18:
Are you using devel or master branch of OctoPiPanel? What version of OctoPrint do you use?

SaftpresseWust, 2016-01-20:
The Keyerrorproblem was solved in devel branch. But still the touch didn't work. As I am not using the adafruit TFT I gave up on this and switched to Touch UI. Even though I liked the simple functional Design of you project.

jonaslorander, 2016-01-21:
Yeah, I've uderstood the touch can be a bit tricky on some displays. You didn't get the touch to work at all, or just nog in OctoPiPanel?

On line 102 and 103 you can change to another lib and device for the touch if you got the touch to work anywhere else but OctoPiPanel. https://github.com/jonaslorander/OctoPiPanel/blob/devel/OctoPiPanel.py#L102

SaftpresseWust, 2016-01-21:
In Raspian the touch is working perfectly. I wouldn't know what to put in your lines 102 and 103. Here is the information about the display I am using http://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B).

jonaslorander, 2016-01-21:
I would suggest you start with running `ls /dev/input/` and see what events you have there and test them out on line 103. Possibly they did not make a link for the touch screen event and called it touchscreen, like Adafruit.

fcawth, 2016-02-13:
Just dealt with this touchscreen issue..  If anyone is using the any of the later development Jessie builds of octoprint, it turns out that the SDL library is does not work correctly with pygame.  The following link will help install the older version.  [https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/pitft-pygame-tips](https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/pitft-pygame-tips)
I had everything working except for the touchscreen, and when I enabled the mouse cursor in the code it was clear that the mouse was jumping all over the place.

jonaslorander, 2016-02-14:
Thanks for the info. Does this solve it wor you as well SoftpresseWust?

HerrRausB, 2016-03-06:
I am experiencing the same problem as SaftpresseWust - not using the original Adafruit PiTFT but this one from german supplier Watterott: http://www.watterott.com/de/RPi-Display-B-Plus

The display works great on my PI 2 using wheezy as current OS release. Also does the touchscreen - X runs flawlessly on the display...

Any new results on that matter? :-)

Regards

Axel

closed 2019-02-03 by TobiasWust