jonaslorander/OctoPiPanel #39: Two issues, both probably very simple.

czyz, 2016-01-24:
One problem that may or may not be a problem -- my pi (using Adafruit's "2015-09-24-raspbian-jessie-pitft28r.zip" image) boots up to an X11 desktop. Do I need to find that in the configuration and make it not launch X or not start a session?

The other problem is that i see a bunch of I/O errors when I launch OctoPiPanel:

sudo python ./OctoPiPanel.py & 
[1] 2538
pi@octopi ~/OctoPiPanel $ sh: echo: I/O error
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
sh: echo: I/O error
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/mode: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/frequency: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/duty: Directory nonexistent
OctoPiPanel initiated
OctoPiPanel started!
---

And I never see any change on the TFT panel. Nothing shows up related to OctoPiPanel.

czyz, 2016-01-25:
Figured out to run raspi-config to set the device to boot into the command-line, so X is no longer taking up the framebuffer. Now OctoPiPanel runs and I see it on the display. Two problems remain - one new:

• The same "Directory nonexistent" messages show up when OctoPiPanel is launched. Perhaps it's trying to use a pin for a different version of the piTFT? The adafruit tutorials mention doing things like this in order to create outlets for commands to control the screen, perhaps that needs to be done for gpio508?:

sudo sh -c "echo 508 > /sys/class/gpio/export"

• The other problem, and this is a minor one, is that while printing OctoPiPanel displays temperatures in celsius followed by the temperature in celsius in parentheses. I'm guessing the line to change to remove the duplicated temperature would be one of these. I don't know python but will probably give it a try:

 # Place temperatures texts
        lblHotEndTemp = self.fntText.render(u'Hot end: {0}\N{DEGREE SIGN}C ({1}\N{DEGREE SIGN}C)'.format(self.HotEndTem
p, self.HotEndTempTarget), 1, (220, 0, 0))
        self.screen.blit(lblHotEndTemp, (self.leftPadding + self.buttonWidth + self.buttonSpace, 60))
        lblBedTemp = self.fntText.render(u'Bed: {0}\N{DEGREE SIGN}C ({1}\N{DEGREE SIGN}C)'.format(self.BedTemp, self.Be
dTempTarget), 1, (66, 100, 255))

czyz, 2016-01-25:
Ah, nevermind the duplicated temperature. Now I realize it's showing the current and set temps. Oops. I'll leave it.

Touches to the screen don't seem to do anything - they work on the X desktop, but not in OctoPiPanel.

jonaslorander, 2016-01-25:
It should be possible to OctoPiPanel while X is running, but I have not tested it so I would say you should not boot into X.

The GPIO messages is there because there are a lot of different displays out there that uses different GPIO pins for controlling the backlight. A better way would of course for OctoPiPanel to check and see if the gpio exists before trying to use it.

You issue with the touch screen might be the same as this:
https://github.com/jonaslorander/OctoPiPanel/issues/38#issuecomment-173476354

On line 103 you should make sure that the touch device event really exist.

fcawth, 2016-02-13:
The touchscreen problem is from a pigame incompatibility with SDL2.x.  There are notes on adafruit's site which I didn't make it to until fighting with the touchscreen issue for a while.  Following their steps to downgrade to SDL1.2, the touchscreen is now working.  [https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/pitft-pygame-tips](https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/pitft-pygame-tips)

jonaslorander, 2016-02-14:
Thanks for the info, I will lok it over. I will not Close this yet, since I would like to tidy up the GPIO issue first.

closed 2021-01-18 by jonaslorander