jonaslorander/OctoPiPanel #40: 3.5" PiTFT Not Full Screen

MilkieDJ, 2016-01-27:
I can't seem to get the AdaFruit 3.5" Resistive Touch Screen to show OctoPiPanel full screen, I have tried editing the config.txt but nothing makes a difference.

jonaslorander, 2016-01-27:
Are you running OctoPiPanel from X or the console? Dev or master branch?

MilkieDJ, 2016-01-27:
I am quite a noob to this, so I'm guessing running from console. I SSH in to start it. No idea if its the Dev or Master Branch, I have no idea what those are. I gave up and got TouchUI working on it instead. I did read somewhere that OctoPiPanel does not resize on screens, but that was dated a year ago.

jonaslorander, 2016-01-28:
DId you see X windows system on the display when bwfore you SSH in to the Pi and start OctoPiPanel?

MilkieDJ, 2016-01-28:
No. Display was blank before starting OctoPiPanel.

jonaslorander, 2016-01-28:
Odd that it should not be in fullscreen then... I have no idea what the problem might be. Which LCD is it that you use, do you have a link?

MilkieDJ, 2016-01-28:
It is the Adafruit PiTFT 3.5" Resistive Touch Screen LCD, link below:
[https://www.adafruit.com/products/2097]

jonaslorander, 2016-01-28:
Did you change the windows_width and windows_height values in OctoPiPanel.cfg file? To the resolution of your screen (480x320).

MilkieDJ, 2016-01-28:
Yes I changed the values in OctoPiPanel.cfg.

jonaslorander, 2016-01-28:
Strange. Unfrotnutalley I don't have access to this LCD my self and can not test it out...

mentaluproar, 2016-02-16:
I have the same problem here.  Is there any way I could help?

jonaslorander, 2016-02-16:
Could any of you take a picture of the problem and attach here?

mentaluproar, 2016-02-16:
http://imgur.com/d8QVHU7

jonaslorander, 2016-02-16:
Thanks. Could you try to comment out https://github.com/jonaslorander/OctoPiPanel/blob/devel/OctoPiPanel.py#L112 and uncomment L113 and L114 and see if that helps?

mentaluproar, 2016-02-19:
Sorry I haven't gotten back to you.  I've been busy lately and haven't had time to try this yet.

mentaluproar, 2016-02-21:
Okay, now I have a bit of time.  I'm just not sure what to do here.  Inserting a hash at the beginning of a line will comment it out, but what next?  How do I use that?  (Sorry for the newbishness)

jonaslorander, 2016-02-25:
Sorry for the delay.

Yes, the pound sign is used to make a line a comment.
Make the Changes and then run OctoPiPanel with ```sudo python ./OctoPiPanel.py```

mentaluproar, 2016-02-29:
I finally had some time to get back to this.  I made the changes and now it looks like this.  Just imagine what could be done with all that extra space. :)
[image]

jonaslorander, 2016-03-03:
Not sure why it looks like this. My guess is still that you have not changed the resolution in the config file. But yu stated earlier you changed it. I will have to look into this more at my end. But I'm on parental leave for the next six month and does not have much time for anything other then my son.

mentaluproar, 2016-03-03:
Congrats

59111, 2016-10-25:
is this issue solved? facing the same problem at the moment.

thanks

jonaslorander, 2016-10-26:
Unknown, but I think this might be related to #53 , and the solution was to switch to devel branch. I still have not got an answer from @MilkieDJ if he/she uses devel or master branch. The master branch does not support to change the resolution.

59111, 2016-10-27:
Changing to devel branch solved the resolution problem. (realized that the official OctoPrint download image includes only master branch of OctoPiPanel)

but now i ended up with a connection Error. When starting OctoPiPanel i get the following output:

pi@octopi:~/OctoPiPanel $ sh: echo: I/O error
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
sh: echo: I/O error
sh: 1: cannot create /sys/class/gpio/gpio508/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio508/value: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/mode: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/frequency: Directory nonexistent
sh: 1: cannot create /sys/class/rpi-pwm/pwm0/duty: Directory nonexistent
OctoPiPanel initiated
OctoPiPanel started!
---
Connection Error (None): None

OctoPrint itself is connected and running.

any idea?

jonaslorander, 2016-11-02:
No idea why you get the connection error and still have OctoPiPanel running... It might be that OctoPiPanel is to fast to try and connect to OctoPrint when you boot up you Pi and OctoPrint has not yet started.

MilkieDJ, 2016-11-21:
I never got this sorted, but would like to! 
I am on the devel branch
I have added window_height/window_width to OctoPiPanel.cfg
I have commented `self.screen = pygame.display.set_mode( (self.win_width, self.win_height) )`
& uncommented 
`#modes = pygame.display.list_modes(16)`
`#self.screen = pygame.display.set_mode(modes[0], FULLSCREEN, 16)`

I also get the same errors while it is running:
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent

Not worried about the errors, but would really like to get it running full screen.

closed 2021-01-18 by jonaslorander