jonaslorander/OctoPiPanel #43: KeDei 3.5 inch 480x320 TFT not working

boson2higgs, 2016-03-09:
i installed my screen with driver take here:
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=124961

screen work great but when i try sudo python ./OctoPiPanel.py &

> File "./OctoPiPanel.py", line 595, in <module>
>     opp = OctoPiPanel(320, 240, "OctoPiPanel!")
>   File "./OctoPiPanel.py", line 135, in __init__
>     pygame.mouse.set_visible(False)
> pygame.error: video system not initialized

jonaslorander, 2016-03-09:
It looks like the scripts you are using sets the frame buffer of the LCD to fb0. You need to change to that frame buffer on this line: https://github.com/jonaslorander/OctoPiPanel/blob/devel/OctoPiPanel.py#L101

Usually fb0 is HDMI output.

boson2higgs, 2016-03-09:
same problem with fb0

jonaslorander, 2016-04-18:
try edint L103 to use event0 instead of touchscreen, or setup a symbolic link in /dev/input pointing to event0.

OmerArad, 2016-07-07:
Hey @boson2higgs, any luck with that? I'm stuck with the same LCD :-\

SydneyMakers, 2016-11-10:
I'm running the Pi 3 with the v6.2 3.5 LCD from Kedei.
switching 
`os.putenv('SDL_FBDEV'      , '/dev/fb1')`
to
`os.putenv('SDL_FBDEV'      , '/dev/fb0')`

And also switching
`os.putenv('SDL_MOUSEDEV', '/dev/input/touchscreen')`
to
`os.putenv('SDL_MOUSEDEV', '/dev/input/event0')`
after confirming it was the touchscreen by running
`cat /dev/input/event0`
and touching the LCD screen.

I've yet to make the OctoPiPanel run upon startup.

closed 2021-01-18 by jonaslorander