jonaslorander/OctoPiPanel #44: 800x480 HDMI+Touchscreen display not working

626Pilot, 2016-04-05:
[This is the display.](http://www.amazon.com/Resistive-interface-compatible-Raspberry-Pi/dp/B00U21UA16?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_detailpage_o04_s00)

I have it set up on a Raspberry Pi 3 running Python 2.7. git pull states that OctoPiPanel is up-to-date. It displays the console correctly (right resolution, no weird bars on the side, etc.) OctoPrint and the webcam interface are both working fine, although it took some hours to set them all up. (The nightly build appears to just be a stock Raspbian Jessie image with OctoPrint, etc. copied in, but not actually set up, or any of the dependencies downloaded.)

On trying to start OctoPanel:
>  sudo python ./OctoPiPanel.py 
> Traceback (most recent call last):
>   File "./OctoPiPanel.py", line 595, in <module>
>     opp = OctoPiPanel(320, 240, "OctoPiPanel!")
>   File "./OctoPiPanel.py", line 135, in __init__
>     pygame.mouse.set_visible(False)
> pygame.error: video system not initialized

I tried setting SDL_FBDEV to /dev/fb0 (which does exist), but that didn't help.

This machine doesn't have X-windows installed. Is that an issue?

jonaslorander, 2016-04-05:
Since you are using the HDMI output you can use the standard settings in Raspbian, and thus you can comment out lines 97 - 103 in OctoPiPanel.py. You make a comment of an entire line by adding a # in the beginning of the line. I just tested this and it seems to be working fine.

626Pilot, 2016-04-06:
These are lines 97 through 103:
        self.buttonWidth = 100
        self.buttonHeight = 25

        # Status flags
        self.HotEndTemp = 0.0
        self.BedTemp = 0.0
        self.HotEndTempTarget = 0.0

I don't think that's what you meant, so I commented this instead:

#        if platform.system() == 'Linux':
#            # Init framebuffer/touchscreen environment variables
#            os.putenv('SDL_VIDEODRIVER', 'fbcon')
#            os.putenv('SDL_FBDEV'      , '/dev/fb0')
#            os.putenv('SDL_MOUSEDRV'   , 'TSLIB')
#            os.putenv('SDL_MOUSEDEV'   , '/dev/input/touchscreen')

I ran the program again. This time, a tiny window showed up on the LCD screen for a couple seconds. Then, the program crashed with this output: $ sudo python ./OctoPiPanel.py 
sh: echo: I/O error
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
OctoPiPanel initiated
OctoPiPanel started!
---
Traceback (most recent call last):
  File "./OctoPiPanel.py", line 596, in <module>
    opp.Start()
  File "./OctoPiPanel.py", line 194, in Start
    self.get_state()
  File "./OctoPiPanel.py", line 293, in get_state
    self.HotEndTemp = state['temps']['tool0']['actual']
KeyError: 'temps'

I have this in OctoPiPanel.cfg:
[settings]
baseurl = http://localhost:5000
apikey = (API key copied from the web UI, a very long hex number)

updatetime = 2000
backlightofftime = 30000

window_width = 800
window_height = 480

OctoPrint is running, connected to the printer, able to cause the heaters to turn on and the machine to home and move around, so all that seems to be okay.

In summary, the window is very tiny even though the window_width|height are set, and it crashes after a couple of seconds. Any help you can give would be greatly appreciated. I'm eventually going to roll all this into a tutorial for getting OctoPrint + OctoPiPanel set up on a Raspberry Pi 3 with this particular screen, and I will credit you for your help.

jonaslorander, 2016-04-06:
You must use the devel-branch of OctoPiPanel. Clone the devel-branch and then follow my previous instructions.

626Pilot, 2016-04-06:
It works perfectly on the devel branch. Thank you very much! :)

closed 2016-04-06 by 626Pilot

626Pilot, 2016-04-09:
After a lot of research into making the touch input work, I discovered that it's not necessary to use the Waveshare drivers at all. The kernel they give you doesn't have any drivers that don't already come with Raspbian! This is particularly good news if you have a Pi Zero or 3, for which no drivers have yet been released.

The touchscreen I linked above is either a [Waveshare 5" (B) LCD touchscreen](http://www.waveshare.com/wiki/5inch_HDMI_LCD_(B)), or a knockoff that uses an identical silkscreen on the board.

[This thread](https://archlinuxarm.org/forum/viewtopic.php?f=60&t=9246) has info on what configuration values to use, and where to put them. Below is a simplified version of that, without any stuff you don't need on a Pi 3. This may also work on a Pi Zero, but I don't have one to test it on. Hopefully, this info will be useful to others who might be searching for a solution. It took hours to find across that thread, which had all the information in one place, including the elusive `gpio_pendown` value.

In /boot/config.txt:
> disable_overscan=1
> framebuffer_width=800
> framebuffer_height=480
> hdmi_force_hotplug=1
> 
> hdmi_group=2
> hdmi_mode=1
> hdmi_mode=87
> hdmi_cvt 800 480 60 6 0 0 0
> 
> max_usb_current=1
> 
> dtparam=i2c_arm=on
> dtparam=i2s=on
> dtparam=spi=on
> 
> dtparam=audio=on
> start_x=1
> gpu_mem=128
> dtparam=spi=on
> dtparam=i2c_arm=on
> dtoverlay=w1-gpio
> 
> device_tree_overlay=w1-gpio-pullup=on
> device_tree_param=gpiopin=4
> device_tree_param=extpullup=1
> 
> device_tree_overlay=ads7846
> device_tree_param=cs=1
> device_tree_param=penirq=25
> device_tree_param=penirq_pull=2
> device_tree_param=speed=50000  
> device_tree_param=keep_vref_on=0
> device_tree_param=pmax=255
> device_tree_param=xohms=150
> device_tree_param=xmin=200 
> device_tree_param=xmax=3900
> device_tree_param=ymin=200 
> device_tree_param=ymax=3900

If you want to test without editing /boot/config.txt and rebooting, you can do this instead:
> sudo modprobe ads7846_device model=7846 speed=100000 cs=1 gpio_pendown=25  keep_vref_on=0 swap_xy=0 pressure_max=255 x_plate_ohms=150 x_min=200 x_max=3900 y_min=200 y_max=3900

However, using the modprobe solution means you have to [download and build](https://www.raspberrypi.org/documentation/linux/kernel/building.md) the RPi Linux kernel, which takes hours to run, and then build (`make; sudo make install`) the ads7846 directory of [fb-tft tools](https://github.com/notro/fbtft_tools). The above-linked thread has advice to put the ads7846_device line into /etc/modules-load.d/raspberrypi.conf. You could also put that into /etc/modules. However, using the device tree overlay option in /boot/config.txt means you don't have to do either. The overlay takes the place of the loadable module.

According to the [fb-tft tools wiki entry for the ADS7846](https://github.com/notro/fbtft_tools/wiki/ads7846_device), you can examine pressure events like this:
`evtest /dev/input/event0 | egrep ABS_PRESSURE` (assuming the touch screen is event0)

To read position events:
`evtest /dev/input/event0 | egrep 'ABS_X|ABS_Y'`

To calibrate the screen:
> sudo -s
> aptitude install libts-bin
> export TSLIB_TSDEVICE=/dev/input/event0
> export TSLIB_FBDEVICE=/dev/fb0
> ts_calibrate

Now you should see the calibration screen, which is mostly black with some text and a crosshair in the top right corner. It will ask you to tap in four corners, then the center. For accuracy, use the stylus or the rounded point of a pen cap. The calibration program will then dump a bunch of numbers to the console, and write them to `/etc/pointercal`.

You can then test the calibration by running `ts_test`. That will let you either drag a pointer around, or draw on the screen. Try drawing a four-letter word to make sure it works right.

Reboot the Pi, then do this:
> sudo -s
> export TSLIB_TSDEVICE=/dev/input/event0
> export TSLIB_FBDEVICE=/dev/fb0
> ts_calibrate

If that works, you're all ready to go. You can run octopi_panel, press buttons, make them work, etc.

jonaslorander, 2016-04-18:
Thank you for this! I have the same screen my self for other uses but have not come around to making the touch work as it should. This will be very helpfull!

PhilMaddox, 2016-04-20:
Hi Came across this thread and am hoping the tutorial is coming along well. Im sure many like myself will be helped by a good tutorial 
Im not very well versed with the in and outs of linux and am struggling a bit with this screen set up.
Ive got it working at the correct res and running Octopi fine i just cant get it to boot to Octopi TouchUI fullscreen. 
I only have the stock Raspbian web browser and no matter what i click it wont go full screen
So far ive been trying on a Pi 2 and the latest Octopi  
Thanks

626Pilot, 2016-04-21:
@PhilMaddox 
If you're running the Raspbian browser, you might want to use that to control OctoPi instead of using OctoPiPanel. OPP is not optimized for bigger displays like these, and you'll get more functionality from just loading the OctoPi page in the browser.

PhilMaddox, 2016-04-21:
@626Pilot 
Thanks for the reply . I have now installed the minimal kiosk browser and followed the instructions>>https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser
seems to work Ok so far.
Kind of being forced to get to grips with Linux isnt a bad thing either

nexus511, 2017-06-26:
Thank you for your detailed description. There is one little thing I want to add: It is important that the audio-signal is not sent to the display via HDMI as this results (at least for my device) in some displacement and colored lines on the top and left of the display.

According the waveshare documentation, one should add
```
hdmi_drive=1
```
to the `config.txt` [[1]](http://www.waveshare.com/wiki/5inch_HDMI_LCD).

That fixes the issue for my device.