jonaslorander/OctoPiPanel #45: stuck getting my 3.5 pitft installed

remarin, 2016-04-09:
PI 2 b
3.5 pitft tft + touchscreen
Octopi .12
Pygame
First, it would be nice if someone wrote a guide from a "Octopipanel for idiots" view point- a detailed step by step guide to get a screen installed for people like me that know nothing about this stuff.  Between getting the pi going and now the display I've wasted a month trying to get everything installed and working.  Information is fragmented all over the place and for me unclear.  Don't mean to project an attitude but I bought the display because it was supposed to be an easy install.

I finally found your installation instructions which have helped but now I'm stuck at the settings step.  Specifically at accessing the octopipanel.cfg file.  I can't find anything on the Web that tells me how to open it.  I got the api key through octopi.local and put my sd card in my laptop thinking I could access it that way but don't see any files or folders for octopipanel.

I'm pretty much at a loss from that point on.  Do I browse to the octopipanel.py file from octopi or from octopipanel?  Automatic startup isn't really clear to me either.

Thank you in advance for any help.  I'd really like to get this thing working.

jonaslorander, 2016-04-18:
All the commands in the readme are supposed to be run using a connection to your Pi with SSH, using for instalnce PuTTY (if you are using Windows) och simply the ssh command (if you are using Linux or Mac OS).

To edit the config file it is a mather of opening the wile, over your SSH connecting with nano (a text editor), ´´´nano OctoPiPanel.cfg´´´ (assuming you are in the same catalog as OctoPiPanel git checkout).

Also, this is only for support with OctoPiPanel, not OctoPi. Perhaps @guysoft could e of more assistance, he has some ready made scripts that update your OctoPiPanel config file with your OctoPrint API-key.

But, what you should be doing firstly is to read about Linux and the command prompt. If you don't even not how to open a text file in Linux you really should be reading up on Linux before coming here complaining about unclear instructions.

closed 2016-04-18 by jonaslorander

remarin, 2016-04-19:
Thanks for the reply but I've already worked my way through editing a file in Linux. 
I've also put in far too much time and effort trying to get this crappy little display to work and threw it out over the weekend. When I buy something that is portrayed as an easy installation only to find out I have to become familiar with an operating system I have no desire to learn I think I have every right to voice my displeasure. 
So thanks again- sorry if I ruffled your feathers. 

Rick