jonaslorander/OctoPiPanel #47: pygame.error: video system not initialized

mrp192, 2016-04-19:
This is what im getting when I try to start OctoPiPanel. Not sure what to do.

 ALSA lib confmisc.c:768:(parse_card) cannot find card                                                                                         '0'
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_card_driver return                                                                                        ed error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_concat returned er                                                                                        ror: No such file or directory
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_refer returned err                                                                                        or: No such file or directory
ALSA lib conf.c:4720:(snd_config_expand) Evaluate error: No such file or directo                                                                                        ry
ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default
Traceback (most recent call last):
  File "./OctoPiPanel.py", line 595, in <module>
    opp = OctoPiPanel(320, 240, "OctoPiPanel!")
  File "./OctoPiPanel.py", line 135, in __init__
    pygame.mouse.set_visible(False)
pygame.error: video system not initialized

jonaslorander, 2016-05-11:
What LCD are you using? How is it connected? Which RPi are you using? Which Raspbian version are you using?

karabas, 2016-06-23:
I got teh same problem with HDMI 7" display (1024x600) and rasp pi3. I solved it by commenting out that block as I did not use X and it seems that second IF string does not do correct.

if platform.system() == 'Linux':
    if subprocess.Popen(["pidof", "X"], stdout=subprocess.PIPE).communicate()[0].strip() == "" :
            # Init framebuffer/touchscreen environment variables
            os.putenv('SDL_VIDEODRIVER', 'fbcon')
            os.putenv('SDL_FBDEV'      , '/dev/fb1')
            os.putenv('SDL_MOUSEDRV'   , 'TSLIB')
            os.putenv('SDL_MOUSEDEV'   , '/dev/input/touchscreen')

jonaslorander, 2016-07-21:
If you run HDMI the framebuffer is not fb1, but fb0.

closed 2016-07-21 by jonaslorander