jonaslorander/OctoPiPanel #48: Compatibility with Raspberry Pi 2?

ChrisWeiss, 2016-05-03:
I had no luck getting the Adafruit PiTFT working on a Raspi 2. My steps were as follows:
Clean OctoPrint install - Expanded disk, set password, region, and time zone.
Reboot
From web UI, install latest (1.2.10).
From SSH: reboot.
from SSH: Run ~/scripts/enable-adafruit-pitft

A new kernel is then compiled (apparently successfully).
When I reboot the Pi, it doesn't get past the rainbow screen, which (from what I can tell) indicates a bad kernel.
Are the new binaries it's trying to compile not Pi2 compatible?

These steps work fine with the same image TFT on a Raspi 1.
Image used: 2015-05-05-octopi-wheezy-0.12.0
Output of enable-adafruit-pitft:
`
pi@octopi ~/scripts $ ./enable-adafruit-pitft

Installing Adafruit PiTFT Display on this installation of OctoPi, do not turn off your Pi during this process!

--2016-05-01 22:09:50--  http://adafru.it/pitftsh
Resolving adafru.it (adafru.it)... 162.243.27.247
Connecting to adafru.it (adafru.it)|162.243.27.247|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://gist.githubusercontent.com/ladyada/f86f777df68fe41181b0/raw/ [following]
--2016-05-01 22:09:51--  https://gist.githubusercontent.com/ladyada/f86f777df68fe41181b0/raw/
Resolving gist.githubusercontent.com (gist.githubusercontent.com)... 199.27.79.133
Connecting to gist.githubusercontent.com (gist.githubusercontent.com)|199.27.79.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14033 (14K) [text/plain]
Saving to: `pitftsh'

100%[=================================================================================================================================================================================================================================================================================>] 14,033      --.-K/s   in 0.001s

2016-05-01 22:09:55 (25.9 MB/s) - `pitftsh' saved [14033/14033]

START
Type = 28r
[ADAFRUIT] Starting image modification
--2016-05-01 22:09:56--  http://adafruit-download.s3.amazonaws.com/libraspberrypi-bin-adafruit.deb
Resolving adafruit-download.s3.amazonaws.com (adafruit-download.s3.amazonaws.com)... 54.231.19.200
Connecting to adafruit-download.s3.amazonaws.com (adafruit-download.s3.amazonaws.com)|54.231.19.200|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 184818 (180K) [application/octet-stream]
Saving to: `libraspberrypi-bin-adafruit.deb'

100%[=================================================================================================================================================================================================================================================================================>] 184,818      316K/s   in 0.6s

2016-05-01 22:09:57 (316 KB/s) - `libraspberrypi-bin-adafruit.deb' saved [184818/184818]

--2016-05-01 22:09:57--  http://adafruit-download.s3.amazonaws.com/libraspberrypi-dev-adafruit.deb
Resolving adafruit-download.s3.amazonaws.com (adafruit-download.s3.amazonaws.com)... 54.231.80.96
Connecting to adafruit-download.s3.amazonaws.com (adafruit-download.s3.amazonaws.com)|54.231.80.96|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 672150 (656K) [application/octet-stream]
Saving to: `libraspberrypi-dev-adafruit.deb'

100%[=================================================================================================================================================================================================================================================================================>] 672,150      504K/s   in 1.3s

2016-05-01 22:09:58 (504 KB/s) - `libraspberrypi-dev-adafruit.deb' saved [672150/672150]

--2016-05-01 22:09:58--  http://adafruit-download.s3.amazonaws.com/libraspberrypi-doc-adafruit.deb
Resolving adafruit-download.s3.amazonaws.com (adafruit-download.s3.amazonaws.com)... 54.231.114.162
Connecting to adafruit-download.s3.amazonaws.com (adafruit-download.s3.amazonaws.com)|54.231.114.162|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 31480442 (30M) [application/octet-stream]
Saving to: `libraspberrypi-doc-adafruit.deb'

100%[=================================================================================================================================================================================================================================================================================>] 31,480,442  2.66M/s   in 13s

2016-05-01 22:10:11 (2.34 MB/s) - `libraspberrypi-doc-adafruit.deb' saved [31480442/31480442]
[... 112 lines cut ...]
Reading state information... Done
Note, selecting 'libts-0.0-0' instead of 'tslib'
libts-0.0-0 is already the newest version.
libts-0.0-0 set to manually installed.
libts-bin is already the newest version.
The following NEW packages will be installed:
  evtest
0 upgraded, 1 newly installed, 0 to remove and 98 not upgraded.
Need to get 21.9 kB of archives.
After this operation, 100 kB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian/ wheezy/main evtest armhf 1:1.30-1 [21.9 kB]
Fetched 21.9 kB in 0s (37.5 kB/s)
Selecting previously unselected package evtest.
(Reading database ... 69319 files and directories currently installed.)
Unpacking evtest (from .../evtest_1%3a1.30-1_armhf.deb) ...
Processing triggers for man-db ...
Setting up evtest (1:1.30-1) ...
Would you like the console to appear on the PiTFT display? [y/n] n
Would you like GPIO #23 to act as a on/off button? [y/n] n
[PITFT] Done

Reboot you Pi to use the Adafruit Pi TFT Display

pi@octopi ~/scripts $ sudo reboot
`

ChrisWeiss, 2016-05-05:
Some follow up - I am able to get the display working on the Pi2 following Adafruit's docs.

ChrisWeiss, 2016-05-05:
Closing as this is an issue with the OctiPi scripts, not OctoPiPanel.

closed 2016-05-05 by ChrisWeiss

jonaslorander, 2016-05-05:
I was about to give that suggestion. There seem to be different kernels for RPi1 and RPi2, which is reasonable since they have different CPUs.