jonaslorander/OctoPiPanel #49: Unable to get Chromium installed to get OctoPiPanel working in a 'kiosk' mode

TimVa, 2016-07-11:
Hello,
I have been following a lot of guides online, but none seem to have a fitting answer to my issues.

Setup : RPI3/RPi 7" Touchscreen/OctoPi 0.13 (Jessie Lite) 
(the real official screen, as I have seen a lot of people being confused about this : [https://www.raspberrypi.org/blog/the-eagerly-awaited-raspberry-pi-display/]

Does it work? Does it work on Rpi3?
A bit more clarity in the 'scripts' might help here also.

Guide followed, but turns bad @ step 4 : https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

Willing to help here, with testing on Rpi2/Rpi3 and RpiTouchscreen7"

Keep up the excellent work
T.

jonaslorander, 2016-07-21:
OctoPiPanel is unable to work in X mode. You can not have X started, as I suspect you have if you have a browser running. If you can get OctoPiPanel running in X it would be appreciated.

yoyojacky, 2016-08-25:
@TimVa  It can be run in Raspberry Pi 3B,  and it works fine.

closed 2021-01-18 by jonaslorander