jonaslorander/OctoPiPanel #51: OctoPi, OctoPiPanel, and Raspberry Pi 3?

CJeffyB, 2016-07-21:
Guy Sheffer suggested I post this over here.  I originally posted it in the OctoPi issue tracker, here:
https://github.com/guysoft/OctoPi/issues/249

I have a Raspberry Pi 3, and have been using it with OctoPi happily for several months. I was recently gifted a PiTFT, and have been completely unable to make it work with OctoPi. Something seems to be hanging on boot, but I have no idea what I need to do next to attempt to track down the problem.

First, I started with my existing install, and ran the ./enable-adafruit-pitft script. Rebooted when prompted, and the device promptly hung. The TFT shows a solid grey-white screen.

I then pulled the Micro SD card, and did a completely fresh install. Ran the script again, and it is hanging in the exact same place.

The system does not respond to anything on the web interface, nor to SSH requests.

Is there something that I'm missing or need to do to make it all work?

Are there any steps that I can take to provide further information to assist troubleshooting?

jonaslorander, 2016-07-21:
Have you tried downloading the latest raspbian (not OctoPi) and running the setup scripts from Adafruit to see if you can get the screen working from there?

https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install#diy-installer-script

I suspect that Adafruit might have updated the setup, procedure form the screen while @guysoft  maybe haven't updated the enable script.

Please try using Adafruits tutorial and see if that works (and let me know the result). If so, you can add OctoPrint and OctoPiPanel to that installation.

CJeffyB, 2016-07-22:
I can definitely give that a shot this weekend.  Before I do, I want to ask...  Do you think I would get better results trying to add OctoPi and OctoPiPanel to a base Raspbian, or would I be better off trying to get the Adafruit instructions to work on a base OctoPi install?

Thanks!

CJeffyB, 2016-07-27:
I've had some free time this week, and have spent the last couple of days working on this.  I started from the most recent stable branch of OctoPi.  I've managed to get the Pi to at least install the display, by following steps 2 and 3 of the instructions on the Adafruit page:
https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install

The Pi safely reboots, and the display shows a (blank) black screen.  I am still unable to get OctoPiPanel to load however.

When I try and run ~/scripts/OctoPiPanel, I get the error:
`
Traceback (most recent call last):
  File "/home/pi/OctoPiPanel/OctoPiPanel.py", line 48, in <module>
    import pygame
ImportError: No module named pygame
`

I followed the instructions over on Adafruit, to make sure that I have pygame installed, and that I have the  correct version of SDL:
https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/pitft-pygame-tips

Do I need to do some other step?  I haven't been able to find ANY documentation on what exactly the scripts in ~/scripts are supposed to do, so I'm hesitant to run the calibrate-rpi-display or enable-rpi-display, as those seem to imply that they set the system up for the official Raspberry Pi display, rather than the Adafruit one.

I'm using a Raspberry Pi 3, with the 2.8" resistive screen, and a Pi camera (v2).

yoyojacky, 2016-08-25:
@CJeffyB  I met this too, you can use this command in your command line to fix it.
sudo apt-get -y install python-pygame

jonaslorander, 2016-08-31:
In OctoPi, the scripts in ~/scripts are used to enable the Adafruit display. But I think Adafruit has some changes to their display drivers and thus these scripts are probably outdated. That is why I suggested you don't use OctoPi but rather install OctoPrint and OctoPiPanel on a fresh Raspbian install where you have ready set up  properly. 

Regarding the missing pygame I think it is installed by default if you use Raspbian (not the lite version). Otherwise you can install it as suggested by @yoyojacky . As stated in the radme of OctoPiPanel, pygame is a required prerequisite!

cruiter1, 2016-10-12:
I followed @yoyojacky advice and it worked like charm. I'm running almost exactly the same hardware and downloaded the latest OctoPi install package with OctoPiPanel presintalled. It looks like the pygame package isn't pre-installed. So along with installing the pitft dyi installed scripts, I had to do pygame also. Then it worked like normal.

closed 2021-01-18 by jonaslorander