jonaslorander/OctoPiPanel #53: Can't adjust resolution an touch doesn't work

tschensie, 2016-08-29:
Hi.
I'm using this 5" Display: http://wiki.52pi.com/index.php/5-Inch-800x480-HDMI-GPIO-Touch-Screen_SKU:EP-0072
I can't setup the resolution for OctoPiPanel. 
I put 
window_width = 800
window_height = 480
into the OctoPiPanel.cfg, but it still starts in a small 320x240 window.
Also the touchscreen doesn't work in OctoPiPanel.
Tried to switch from Touchscreen to event0, in OctoPiPanel.py no success.

I'm using a Pi3 with the OctoPi Image.
When deactivating OctoPiPanel and booting to the gui I have the right resolution (800x480) and the touch also works.
How do I have to setup OctoPiPanel to work ?
Thx.

tschensie, 2016-08-29:
Little Update: After reading the issue #52 I saw that may touchscreen ist also on event2.
So I changed it in octopipanel.py with the result, that octopipanel disappears after pressing the touch an the commandline ist shown....

tschensie, 2016-08-29:
ok, next update: Touch seems to work now.
I don't have a printer connected yet, because it insn't build ready yet.
when I press reboot or shutdown on the touch octopipanel ends and the commandline is shown.
when i tab somewhere else on the screen nothing happens.

It seems I only got the problem to set the resolution to 800x480.

jonaslorander, 2016-08-31:
You have to be connected (the printer state in OctoPrint has to be at least connected) to a printer for OctoPiPanel to work, otherwise it will crash (I guess) when trying to send commands. It should be enough if you just connect you printers control board and make sure OctoPrint can make a real connection.

Regarding the resolution, are you running the dev branch or master? Changing of resolution is not implemented in master branch.

yoyojacky, 2016-09-02:
you should  use command like this to get devel branch ： ” git clone  URL   -b devel   "

yoyojacky, 2016-09-02:
hi tschensie, I suggest you modify  /boot/config.txt file and change it like this:
framebuffer_width=800
framebuffer_height=480
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt  800 480  60  6  0 0 0 
save it and reboot it ..

yoyojacky, 2016-09-02:
But I still have a problem that i can  see the octopipanel screen, it's none  response when i touch the button..
I have tried to use evtest to check my touch screen ,it works properly... 
Is there any solution that can fix it？

tschensie, 2016-09-02:
@jonaslorander I'm ruinning master branch. Thought it would work because it's descriped in the the settings for different resolotions (no hint for dev branch). I will change to dev and test.

@yoyojacky thanks, but no need for this. l got it in the config.txt, otherwise I would not get the right resolution when booting to gui like described in first post.

tschensie, 2016-09-02:
@yoyojacky try ls -l /dev/input/* and look where to find your touch. My is on event2. Enter this OctoPiPanel.py and it should work.

jonaslorander, 2016-09-15:
The default branch when browsing OctoPiPanel on GitHub is devel. As you can see on the branch change drop down.

tschensie, 2016-09-15:
Hmm, tried to sitch do devel.
when I enter "git checkout devel" I get "error: pathspec 'devel' did not match any file(s) known to git."

tschensie, 2016-09-16:
Ok, it works :-)
Could not switch to devel, so I deleted the whole OctoPiPanel folder an installed it again.
Put the correct values to OctoPiPanel.py (fb0 and event2) and the correct resolution to octopipanel.cfg, reboot and it works.
Octoprint ist shown at 800x480 and touch works like charm (no printer connected, but reboot and shutdown works)
Thx for help.

closed 2016-09-16 by jonaslorander