jonaslorander/OctoPiPanel #54: pygame.error: video system not initialized and touch not working

Aiddy81, 2017-01-09:
Hey,

I had an issue on my RPi 1 Model B w/ Adafruit resistive touch TFT panel starting OctoPiPanel from my OctoPrint installation. After spending a few hours trying various things I stumbled across this forum https://www.raspberrypi.org/forums/viewtopic.php?p=527172 the last post was rather helpful and I found if I changed touchscreen in this line 

os.putenv('SDL_MOUSEDEV' , '/dev/input/touchscreen')

to event0, event1, mouse0 etc, I was able to get the OctoPiPanel to display on the 2.8" TFT but now unfortunately I can't seem to get the touch part to work. Just wondering if you have any advice?

jonaslorander, 2017-01-10:
os.putenv('SDL_MOUSEDEV' , '/dev/input/touchscreen') this line tells Python what touchscreen controller event to use. If you do `ls /dev/input` what events do you get? Perhaps the touchscreen symlink is not setup.

If you are using the adafruit resistive tft you should be able to install in according to the README file of OctoPiPanel and everything should work. This is the same setup I have developed OctoPiPanel on.

Aiddy81, 2017-01-10:
Hi Jonas,

Thanks the reply and assistance, it's very much appreciated, when running ls /dev/input only the below items show:

- by-id
- by-path
- event0
- event1
-  mice  
- mouse0

I've also followed this https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/touchscreen-install-and-calibrate and can tick the boxes regarding the STMPE rules and running `sudo rmmod stmpe_ts; sudo modprobe stmpe_ts,` so not too sure id I've missed something.

Aiddy81, 2017-01-22:
Hi Jonas,

Just an update on this after a spending some hours diagnosing this I also connected the manufacturer of the TFT (in this case Adafruit), it turns out its a problem with the screen. Here is a link to the forum discussion.

[](https://forums.adafruit.com/viewtopic.php?f=50&t=109637&p=548864#p548864 )

jonaslorander, 2017-01-23:
To bad about the faulty Pitt.  Hope it work better with the replacement one! Please keep me updated,  and if it works you can close this issue.

closed 2020-05-15 by Aiddy81