jonaslorander/OctoPiPanel #55: Does't seem to even be connected

JetLennit, 2017-01-16:
I can't seem to find out how to fix the problem I'm having. After going through the installation process I turned on my Pi and it seemed to work, but I soon found out that the screen wasn't really showing any values. All the temps are 0º and I can't control the printer. It is almost like its not even connected, but I don't know how to connect it. I regularly run prints from OctoPi in-browser so that is connected. Did I miss a step during installation? Am I just dumb and missing something obvious? Any help would be appreciated. I am running it off of a RPi 2B (I think) with a 2.8 inch TFT if that helps. 

Thanks,
Jet

closed 2021-01-17 by JetLennit

guysoft, 2021-01-18:
Worth noting this project is abandoned.

jonaslorander, 2021-01-18:
I thought I had archived the project. I will do it now.