jonaslorander/OctoPiPanel #56: GPIO Error

RobDrizzle, 2017-02-20:
I have the Adafruit 2.8 resistive touch screen (built verbatim to the instructions) and I get
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
errors at startup and then a repeating
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
error for the duration of operation. The console is painfully slow and virtually unusable.

How can I correct this?

BTW love the idea and can't wait for this to work!

jonaslorander, 2017-02-20:
These error are because the code has been adapted to several different screens, and they all use different GPIOs to control the backlight. https://github.com/jonaslorander/OctoPiPanel/issues/39#issuecomment-174411915

If you want to get rid of them, comment out the ones not used by your screen. In OctoPiPanel.py in devel, look at lines 152-160, 184-186 and 199-201. Eg., the ones that you get an error about.

RobDrizzle, 2017-02-20:
Thank you sir... I do have a follow up question. In OctoPiPanel.py, the GPIO pins are labeled 252, 508, etc. I have always seen the Pi's pins as 24, 25, etc. I am thinking that the Backlight pin is on either 24 or 25, so can I just simply replace 252 with 24?

Thank you for the quick turn-around!

jonaslorander, 2017-02-21:
THe backlight on Adafruit 2.8" resistive is on a GPIO of the LCD controller chip. The Adafruit driver sets it up as a virtual GPIO in the `/sys/class/gpio/`-path called gpio508.

It does also use GPIO18, but you have to make sure you are not driving anything on GPIO508 first.

https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/backlight-control

closed 2021-01-18 by jonaslorander